AgonPlatform/agon-mos #2: MOS modules?

stevesims, 2023-08-17:
MOS is currently a monolith.  The only concession to this is support for executable commands as files stored inside the `mos` folder at the root of the SD card.

The Acorn MOS of the BBC Micro supported additional ROMs that could add additional functionality.

Arthur, and later RISC OS, included support for "Relocatable Modules" which could be soft-loaded from disc or present in ROM as an equivalent mechanism for extending the OS.

Ideally MOS should support a module system to allow it to be extended.

stevesims, 2023-08-17:
A module mechanism like this could be used to allow for joystick port support.

Different joystick interfaces could provide modules supporting the same software API, allowing software to be written that can easily target all types of joystick interface.

sijnstra, 2023-10-26:
Interesting idea although I'm not sure of the extent of modularity and what that would mean. As a design principle, I'm not keen on losing memory space for additional modules as the default mode of operation. If the devices are interrupt driven then you could just update the interrupt vector for the respective driver? Is that the level you are looking for - a known API to select specific interrupt so that they can move around between MOS versions without needing to know the exact location of the driver code? Does that meet the driver requirement?
In a way, I see the MOS more as a monitor than an OS - a boot strap to get you off the ground, but if you want OS functionality, there are options to load on top like CP/M, TRS-OS and ZealOS (8 bit).

stevesims, 2023-10-26:
yup - it's all tricky, and knowing how best to deal with this kind of thing is _hard_ :grin:

My (incomplete) understanding of how Acorn did this kind of thing was thru "vectors" which were _kinda_ like interrupts, but not necessarily always interrupt driven.  The mechanism they used _feels_ sensible.  At this time I'm not quite sure how to bring that kind of functionality to the agon-mos, and need to do some research and experimentation to work that out.

I'm not keen on losing memory either.  I wonder if it's possible to have such driver modules optionally stored in flash on the eZ80, and thus not taking up RAM space.

As I see it, MOS needs to be more than just a BIOS or HAL, and I can't envisage a time when to actually use your Agon you'll _need_ to load an OS on top of it.  Sure, you can run a more fully featured OS on top of it, but to me that is akin to running Windows or GEM on top of DOS.  I think it's an important feature of the Agon platform that you can plug in an Agon machine and actually use it without having to load anything else in.

sijnstra, 2023-10-31:
One way to do it is an extensible device control table (DCT), which of course takes up some memory and needs laying out properly (TRS-DOS/LS-DOS for the TRS-80 model 4 is a good example), but if you have enough of the drivers in flash with some other kind of de-referenced hook, you could point to them by something like "request location of built-in mouse driver 2" (allowing to BYO if you want), then "do action on the generic mouse driver" which in this case would be "install address I just received as a mouse driver", could be get coordinates etc - the driver would need to play nice and hook in to interrupts if desired or not in the case of joystick . Wouldn't take up too much memory I guess? Could allow for testing as well?

stevesims, 2024-12-02:
This idea was also posted as an issue to Quark MOS here breakintoprogram/agon-mos#91

There is some interesting (further) discussion about the idea there, including an example of code relocation from MSXDOS in Z80 mode from @S0urceror

stevesims, 2025-04-06:
Now that MOS 3.0 beta 1 has been released, I think it's worth revisiting the overall concept of "MOS modules" - and what exactly that might mean.

MOS has grown; the MOS 3.0 beta 1 build only leaves just over 24kb of free ROM space.  There are still lots of ideas for new features to add into later versions of MOS, and little space to squeeze them into.

A full "relocatable module" system (like RISC OS) where modules can be loaded and remain resident in memory is probably not a good fit for the Agon platform.  We just don't have the free RAM, and so would likely create compatibility issues with existing software.  It would also be very difficult to implement.

A system of transient modules, like ROMs that can be paged in and out as on the BBC Micro, may be a better model, and a more practical one to achieve.

Such transient modules could be loaded into the moslet memory space.  A single module could, potentially, implement multiple star commands, and/or multiple API calls.

Using the moslet memory space would make it tricky to make these modules compatible with existing moslets.  Similar issues would exist with a module attempting to use a command/API from another module.  It may be possible to allow this - it just wouldn't be quick, as there would be a memory swapping process that would need to happen between calling APIs/commands.

Looking beyond MOS 3.0, some new features such as Unified I/O #80 almost certainly have to be built into the code OS functionality, and always resident in ROM.

A feature such as the evaluation engine #73 on the other hand would be far less likely to need to be always resident.  An API call to `mos_evaluateexpression`, or a use of `*seteval` or `*if` could load the module on demand.

If we implement such a module system we should consider which potential new MOS 3.x features don't really need to be "core" and could be be grouped together into a single module.  For example, ideas around enhanced keyboard and joypad APIs could use this (see #106, #107 and #108).  It may also be worth considering whether parts of an enhanced/modular line editor could be split out too - for example tab-completion arguably need not be built-in and could be a module.

There may also be scope in moving some other parts of MOS into modules.  The star commands that handle variable management (`set`, `setmacro`, and `show`) could move to a module, with their APIs remaining resident in ROM.

There may also be compatibility issues with large programs that encroach into moslet space.  That may affect which existing MOS features (commands/APIs) we can consider moving out into modules, and which must remain resident in the ROM.  As a general rule, MOS 2 features would probably need to remain in ROM for compatibility.

stevesims, 2025-04-06:
There are potential issues with this modules idea around programs that use memory inside the moslet area, whether that is existing programs, or ones yet to be written.

Detecting whether an existing program is using moslet memory when an API call is made would be hard, and most likely impractical for various reasons.

A simple solution is to just not support module-based APIs/commands when running existing programs.

We could extend the executable header to support one or two flags to control whether and how module-based APIs/commands are used.

One flag could be a blanket/simple "enable modules".  This would be a contract that states that the program does not, and will not, use the moslet memory space.

A second flag could indicate that moslet space _is_ being used, but that it is safe to swap that space out to disk and restore it after a module-based API call is made.  A program setting this flag would be making a contract to say that it won't pass in pointers to buffers that overlap the moslet area when making module-based API calls.

It may be safest for MOS to always assume on program exit that the moslet/module space is invalid, whether flags have been set or not.

Module-based APIs should probably check pointers to buffers to see if they overlap the moslet/module memory area, and if so the API call should fail.  "Core MOS" APIs, present in the ROM, would be able to accept pointers to buffers that overlap the moslet memory area.

We may wish to consider exposing APIs for malloc/free for modules to use, as having them include their own malloc/free implementations would be expensive.  These would represent a potential memory leak risk though.

stevesims, 2025-04-09:
As well as APIs and commands, modules could also provide support for I/O streams if/when we add support for unified I/O #80 

This could be how we add support for the `usbserial:` stream #157 and potential other streams too

stevesims, 2025-04-09:
Another potential item to put in a module would be help text. This could be help that goes beyond help for commands. RISC OS would let you get help for a few things besides commands…

There could be other things to put in modules as well - perhaps C functions, as opposed to APIs. There may be other things too not yet thought of 

The implication of this is we should have an extensible module format that can potentially support many different types of content. A header that is easily parsed to find matching content when needed

Thought will be needed about what info to cache in advance, and what to scan for on demand

There may be potential to allow this module mechanism to be used by applications to load code into different memory locations. That too needs more thought

stevesims, 2025-04-11:
Idea: modules could potentially include within them a command string/script to perform initialisation functions.

A command could be added to MOS to "scan for modules".  This would likely search for modules in the `moslet$path` and gather information from them.  Scripts within modules could potential set up system variables that the module may need.  A system variable or two could be set when calling these scripts to give the script context - this me be the directory in which the module is located, the module filename, possibly the complete module file path.

We could also consider using this mechanism for registering module features, perhaps adding star commands to "register API" and "register command".  This would be an alternate approach to scanning on demand.  It would however potentially place a heavier memory burden on MOS for keeping track of this kind of information.