AgonPlatform/agon-mos #3: Calls to check capabilities

stevesims, 2023-08-17:
Ideally MOS should have API calls to allow for software to check for the capabilities of the machine on which it is running.

A Console8 for instance has joystick ports which may not be present on other Agon Light machines.

If software can interrogate the MOS about the capabilities of the machine then it can intelligently offer choices to the user.

julianregel, 2023-08-30:
This sounds like a useful feature! In reading this, I was reminded of a couple of comments made by JGH over in Stardot (https://stardot.org.uk/forums/viewtopic.php?p=395021#p395021):

> In general, you should not interrogate the OS to decide what facilities are present. You should test to see if the facilities themselves are present.
>
>Wrong:
>IF NOT Master THEN no real-time-clock
>Correct:
>IF OSWORD 14 returns the real-time-clock THEN there is a real-time-clock
>
>Wrong:
>IF NOT Master THEN no configuration memory
>Correct:
>IF OSBYTE 162 returns null value THEN no configuration memory
>
>etc.

And more recently (https://stardot.org.uk/forums/viewtopic.php?p=400748#p400748):

>The only way to deteramine if *ANYTHING* is present, is to attempt to use it and not what happens. You test for the presense of a real-time clock by trying to read the real-time clock, not by making assumptions based on other things. You test for the presense of a filing system by trying to select the filing system, not by making assumptions.

IMHO, this seems like a very sensible approach, especially given the open source nature of the Agon platform.

stevesims, 2024-10-24:
more recent versions of MOS have added a "not implemented" return code.  since this issue was written, any attempt to use a MOS API call that didn't have an implementation would return that code.

closing this issue, as it's now effectively obsolete

the Agon-docs repo should provide advice on "feature detection" - see agonconsole8/agon-docs#51

closed 2024-10-24 by stevesims