AgonPlatform/agon-mos #24: Don't push repeated commands to CLI history

stevesims, 2023-10-31:
If the last command entered is a repeat of the one before, don't push it to history

Also consider - if the last command entered was previously entered, then _maybe_ move that command to last history entry instead of pushing a copy?

badre2911, 2024-01-17:
I suppose that the history is put in a table (array), you just have to see if it already exists in this table, if it does not exist we do not insert it in the history table.

stevesims, 2024-10-24:
version 2.3.0 implements the "If the last command entered is a repeat of the one before, don't push it to history" idea.

will close this for now, as the basic feature has been added.  if further refinement to how the history system works we should raise a new issue

closed 2024-10-24 by stevesims