AgonPlatform/agon-mos #25: Suggest to improve the HELP command.

lennart-benschop, 2023-11-05:
If you type HELP at the mos prompt with no further parameters, you get: ERRROR '' not found. This is not very..... uhm... helpful? It could output  the output for HELP HELP instead (suggest to type HELP <cmd> or HELP all) and then give a list of commands.

stevesims, 2023-11-05:
doh - yeah - that's a bug - it _should_ really be showing the `help help` output

stevesims, 2023-11-12:
ok - Dean had _partially_ fixed it, so you no longer get a corrupted error report - instead `help` without an argument _currently_ comes up with `Invalid command`.  so there's still a bug to fix here :grin:

stevesims, 2024-03-30:
this bug is now properly fixed, and help is now working properly as of 2.2.0

the new help command automatically works out the list of available commands, and will automatically work out aliases too