AgonPlatform/agon-mos #32: Adding support for new layouts supported by vdp-gl

GaiusJocundus, 2023-11-19:
The following change is needed to be able to select the new layouts (including dvorak) in agon-mos:

```
diff --git a/src/mos.c b/src/mos.c
index 8559922..c0e9fc8 100644
--- a/src/mos.c
+++ b/src/mos.c
@@ -637,7 +637,7 @@ int mos_cmdSET(char * ptr) {
        ) {
                return 19; // Bad Parameter
        }
-   if(strcmp(command, "KEYBOARD") == 0 && value <= 8) {
+ if(strcmp(command, "KEYBOARD") == 0 && value <= 18) {
                putch(23);
                putch(0);
                putch(VDP_keycode);
```

Otherwise we get an "Invalid Parameter" error even when selecting supported layouts.

Related issues: 

- https://github.com/AgonConsole8/agon-vdp/issues/99
- https://github.com/avalonbits/vdp-gl/issues/33

GaiusJocundus, 2023-11-19:
Relevant PR: https://github.com/AgonConsole8/agon-mos/pull/33

stevesims, 2024-03-30:
all present and future keyboards supported by the VDP can now be selected from MOS as of the 2.2.0 release

MOS has hard-coded help text, which has been updated to the current keyboard list, as of March 2024.

closed 2024-03-30 by stevesims