AgonPlatform/agon-mos #42: Need more file system syscalls, in particular opendir/readdir

lennart-benschop, 2024-03-23:
Apparently it is still not possible for a program to obtain the list of files in a directory.  We can call the CAT command but this prints the directory on the screen. If we need to get the list of files in a program,. so we can make our own file load/save dialogue or Midnight Commander clone we are SOL.

I will  try to implement this.

lennart-benschop, 2024-03-24:
There are system calls defined in mos_api.inc, ffs_dopen, ffs_dclose, ffs_dread, but they are NOT implemented currently. Implementing seems very straightforward.

The only thing is: they need DIR and FILINFO structures. For the DIR struct your application only  needs to reserve a memory area of the appropriate size, for the FILINFO struct, it is fairly clear what goes inside it and this contains the very stuff (file size, data, file name, attribs) that your application wants to read from the directory.

An alternative could be to statically reserve a single DIR struct and allow for one directory to be open at a time and then have mos_opendir/closedir/readdir functions to work on that single DIR struct.  For now I think I will go for the simple straightforward fs_ functions that require applications to reserve their own DIR struct.

I only consider it done if I have an example program that lists a directory based on these system calls.

HeathenUK, 2024-03-24:
@tomm implemented these calls in his "Rainbow" MOS build. [Here's the commit.](https://github.com/tomm/agon-mos/commit/8dd04b45757565d63ede89ac9d05faf19a14f8c9)

This is a simple shim, but since all three functions work via being passed a reference to a DIR/FILINFO struct there's no need to implement anything more. The user can do it themselves.

I'm using these functions happily enough in AgDev now, and if/when they ever get mainstreamed I will offer up the changes to mos_api.h and mos_api.src that match Tom's shim.

stevesims, 2024-03-30:
the new calls implemented in @tomm 's "rainbow" MOS build have been integrated and released as part of the 2.2.0 release

there are however still several filesystem sys calls that are missing

stevesims, 2024-10-24:
Closing as the directory management functions are now exposed

I have raised a separate issue #92 to track the implementation of other fatfs APIs

closed 2024-10-24 by stevesims