AgonPlatform/agon-mos #52: Feature Request: Ignore "System Volume Information" folder

eightbitswide, 2024-04-01:
Despite trying every posted solution to preventing Windows from creating a "**System Volume Information**" folder, mine is instant on creating this folder every time my Agon SD card is presented to Windows.  _(I know, Ditch this crap OS and go back to Linux!)_

Any chance we could have Agon's directory command skip listing this folder?  It's a hidden folder and generally out of view from everything except my Agon which is pleased to allow it to reformat my directory listings in MODE 12 since it's so long.

The option would be to Ignore hidden folders maybe?

Jeff

stevesims, 2024-04-01:
currently our directory command is showing _all_ files it finds

fatfs however does provide attributes data, which we are already using to see that an entry within a directory is a subdirectory.  attributes can also have an `AM_HID` value to indicate that the file is "hidden" and should not be shown in a normal directory view.  additionally there is an `AM_SYS` attribute for "system" entries - which I strongly suspect that would also apply to the "System Volume Information" folder.

what we should look to do is add support for an "a" flag on directory listings to show "all" entries, as a companion to our new "l" flag

we could also look to supplement our directory listing to reveal more of these flags, plus look to add commands that potentially allow adjusting of these attributes, as appropriate.  this might just be setting files as "read only"

info on fatfs's file info object is here: http://elm-chan.org/fsw/ff/doc/sfileinfo.html

stevesims, 2024-11-30:
MOS 3 alpha release includes a new version of the DIR command that, by default, will not show hidden or system files.  there are now flags `-a` and `-s` to show those files

closed 2024-11-30 by stevesims