AgonPlatform/agon-mos #66: fread and fwrite works only with ADL pointers

nihirash, 2024-04-17:
I can open file by 16 bit pointer in legacy cpu mode but when I'll try read or write to buffer from non-ADL mode I'm getting failure - cause API works only with full ADL pointers.

As current workaround I'm using prefixed LD command(LD.LIL) but best solution will be making API consistent and allow use fread and fwrite with non-ADL pointers(when MB isn't equal zero).

calc84maniac, 2024-04-17:
I'd also like to add to this that `ffs_api_fread` and `ffs_api_fwrite` unnecessarily destroy a shadow register which isn't really noted in the API (and no other APIs, or anything in MOS in general, seem to do this).

This code sequence:
```
			EXX		
			LD	HL, _scratchpad
			PUSH	HL
			EXX
```
Could be replaced with this sequence, which is also smaller:
```
			PUSH	HL
			LD	HL, _scratchpad
			EX	(SP), HL
```

stevesims, 2024-11-29:
These fixes got back-ported, and released as part of MOS 2.3.2

closed 2024-11-29 by stevesims