AgonPlatform/agon-mos #107: More sophisticated keyboard API

stevesims, 2024-12-02:
We have three API calls in MOS for direct keyboard interactions.  Firstly there's the simplistic `mos_getkey` which is a simplistic call to wait for a key to be pressed returning the key value, then we have `mos_getkbmap` which provides access to the keyboard map object which gives full info on which keys current are (or are not) pressed, and finally there is `mos_setkbvector` which lets programs directly interpret keyboard data packets sent by the VDP.  (The fourth API we have that deals with the keyboard is the line editor.)

On the foundation of these three APIs it is possible to build programs that deal with any kind of direct keyboard interactions, but they're not the friendliest of interfaces.

The primary need for direct keyboard interactions is for games which tend to need to detect when the user is pressing key that corresponds to a direction ("up", "down", "left", "right"), or "fire".  They may also need to work out when multiple keys are pressed, and when the user has _released_ a key.

It is slightly awkward to do this with any of the existing APIs.

An API that effectively provides a relatively simple wrapper on top of `mos_getkbmap` may be a useful addition.

stevesims, 2024-12-20:
we'd likely want an API to work out where in the kbmap data block a particular key lives.  this may mean providing an API to give the offset and bit for a virtual keycode, and/or an ASCII value.  this may be in addition to an API that just says "is key <x> pressed"

sijnstra, 2024-12-20:
Can this API also solve the hold-shift while rebooting to prevent load of autoexec.txt?

stevesims, 2024-12-20:
autoexec prevention is already present in the MOS 3 branch if you're running a VDP with the required support

sijnstra, 2024-12-21:
Apologies! Your amazing work is bringing features faster than I can keep up with...