AgonPlatform/agon-mos #110: Improve handling of unrecognised VDP protocol messages

stevesims, 2024-12-04:
In principle, the VDP protocol handler inside MOS should reject protocol messages it does not understand.

Unfortunately the implementation of this is simplistic.  On encountering a message type byte that is not within the "understood" list of messages, it _immediately_ aborts the protocol packet handling.  The next bytes received will therefore be attempted to be interpreted as message type bytes.

A VDP protocol packet has a relatively simple design - one byte is the message type, always with the top-bit set, the second byte is the length of the data block, and then following that will be all of the data bytes.

What the protocol handler _should_ do on receiving an unrecognised message type is to read the length and then discard that number of following bytes.

HeathenUK, 2024-12-04:
Good idea! Are there other instances beyond an unknown message type where it would be sensible for MOS to bail intelligently like this?

stevesims, 2024-12-04:
yes - as can be seen in my PR, the protocol handler had two scenarios where it would just bail out.  the first is when it's a message that's not of a type that's (currently) understood.  the second is on receipt of a packet where the "length" byte exceeds the protocol buffer size

in both cases the code in the PR should now discard the entire protocol packet

the changes make MOS safer when encountering packets it does not understand.  in my testing for this I had the main loop of the VDP periodically send bytes for an unknown command with a length 6 bytes, and had the _data_ block for that packet be a valid "keycode" packet for pressing the letter `A`.  before the changes in the PR MOS would behave as if the `A` key were being pressed.  after the changes the unrecognised packet was successfully ignored.

the changes in the PR assume that protocol packets are valid, i.e. they are in the form of a "control" byte, followed by a "length" byte, and then data bytes for that length.  this of course requires the VDP to be well behaved, and only send valid VDP protocol packets, but that seems like a reasonable assumption to be making

stevesims, 2024-12-05:
this is implemented in the MOS 3 branch

#111 includes this fix for MOS 2

PR #111 (2024-12-07): improve how VDP Protocol ignores packets
changes the VDP protocol handler code so that it will always consume the entire protocol packet, whether or not it is of a type that MOS understands

additionally changes the handler to properly discard packets that are too long for the configured buffer. in future this behaviour may be changed to allocate a buffer in the heap of the required length

this fixes an issue where receipt of a packet that MOS does _not_ understand would ignore only the control/type byte, meaning that the data block for that packet could get interpretted as a different packet [... cut]
closed 2024-12-07 by stevesims