AgonPlatform/agon-mos #118: i2c read instructions fail for some devices

richardturnnidge, 2024-12-10:
I have managed to communicate with many i2c devices, see my videos:

https://www.youtube.com/watch?v=WQATfp2rR_Y&list=PL-WZxPxo1iaBRPJj_mS7feUh_gO6pn-Qe

However, I have failed to get a few devices to communicate with Agon. One example is the MPR121 touch sensor.
I get no errors from the MOS commands, the device is seen on the i2c bus, but no data (or perhaps just 0) is returned when data is requested. MOS fills the required buffer with 0s instead of the expected data

**Background**

For most simple devices, you write a _command_, followed by a read request.

The _command_ is normally just the device's register you want to write to or read from.

So, you might write the byte 12 first, then when you read, you can read 1 or more bytes starting from register 12.

The bytes read end up in the buffer pointed to by HL.

For all of the devices in my videos, this works perfectly.

However, a few devices have not sent back the required data.

**Problem**
I _suspect_ that the problem is that some devices expect the initial write to be followed immediately by the read instruction, without a break.
If we send a MOS write, followed by a MOS read, the 'STOP' code is included in between the two. Therefore, the command is not complected by the device.

In MOS we have just two instructions, write, and read. We do not have a 'read from register x' which could combine the two.
See p. 25 of the touch capacitor notes.

In MOS we have just two instructions, write, and read. We do not have a 'read from register x' which could combine the two.

envenomator, 2024-12-11:
Taking a guess here, that this device behaves a little different compared to most other I2C devices, in the sense that it expects a read command byte, followed by a RESTART. Most devices can do a read after a STOP sequence, but it looks like the MPR121 does that a little different.
Someone would need to have access to this device while observing it's behavior using a protocol analyzer, before any (if any) changes can be made to the driver state machine.

richardturnnidge, 2024-12-11:
Thanks for your thoughts. You are correct, all the other devices in my videos worked fine with a STOP between the _write_ and the _read_.

The other module I was trying was a very simple ambient light sensor. Again, worked fine on Arduino, but not on Agon. After many hours trying, I put it down to the same issue.

[image]

[veml7700 light sensor.pdf](https://github.com/user-attachments/files/18095169/veml7700.light.sensor.pdf)

envenomator, 2024-12-11:
If we can somehow confirm this to be the case, preferably by analyzing the behavior over the line, a potential solution could be to create another API call next to the (currently very simple) write/read calls, combined with a I2C state-machine change, where the call flags a RESTART start to be applied during it's invocation.

richardturnnidge, 2024-12-11:
Given that every device I have used (so far) needs a _write register_ before reading, to say what register you want to start reading from, it makes sense to have an api call which does the two things in one.

eg:
  0x??: mos_i2c_read+
  Read a block of bytes from the I2C bus (Requires MOS 1.04 RC3 or above), starting at register A.
  
  Parameters:
  
  A: Register to start reading from
  C: I2C Address
  B: Number of bytes to read (maximum 32)
  HL(U): Pointer to a buffer to write the bytes to

However, most devices have been ok with a _write_, then a separate _read_. Whether they would work ok with a joint command, I don't know.