AgonPlatform/agon-mos #162: Expose low-level SD card functions as MOS APIs

stevesims, 2025-04-11:
It has been suggested by @tomm on Discord that exposing the low-level SD card functions that FatFS uses for SD card access would be a useful addition.

The specific functions, as defined in `sd.h` and implemented in `sd.asm` are these:

```
BYTE	SD_readBlocks(DWORD addr, BYTE *buf, WORD count);
BYTE	SD_writeBlocks(DWORD addr, BYTE *buf, WORD count);

BYTE	SD_init();
```

> That would allow other OSes to read the partition table and provide low level block access for their own FS, without rolling their own sdcard access necessarily

A potential slight complication in exposing the read/write functions as an API is how best to handle the `DWORD addr` argument, as that is a 32-bit value.  The Zilog C calling convention essentially means this 32-bit value should be pushed to the stack in 6 congituous bytes in little-endian order, where the final two bytes will be ignored.  NB this `addr` value is not a memory address, but a disc sector address.

There are three ways this could be done.

The most simplistic way is to use two 24-bit eZ80 registers, using all 3 bytes from one register, and the just low byte from the other for the most significant byte of the 32-bit value and just push them both in the appropriate order to the stack.  This is essentially the approach taken by the `mos_flseek` API which takes a 32-bit "offset" value passed in registers HLU and E.  The implementation of this API performs a `PUSH DE` followed by a `PUSH HL` to push these values.  Essentially the same approach is used for the `ffs_flseek` API, using different registers.

The down-side of this approach is that as it uses all three bytes of HLU it is not compatible with plain Z80 code, as in Z80 mode the upper byte of HLU essentially does not exist.

(The `ffs_ftell` and `ffs_fsize` APIs also uses this approach to return a 32-bit value, using DEU and C.)

An alternate approach for the API would be to use a single register with a pointer to a 4-byte 32-bit `addr` value.  As the value itself is not passed in registers, but instead a pointer is, an API using this approach can be compatible with calls from plain Z80 code.  This approach is used as a way to _return_ 32-bit values by the `ffs_getfree` and `ffs_getlabel` APIs.

(In hindsight, it may have been better if the `ffs_ftell` and `ffs_fsize` APIs had accepted a pointer to a 4-byte block of memory as a parameter, and stored the value the C function they call into that block of memory.  Perhaps alternate Z80-friendly APIs could be added for those calls, and also for the two `*_flseek` APIs).

One final approach could be to use two registers, but only use the lower 16-bits in each.  This would also provide an API that is compatible with both Z80 and eZ80 code, but it would arguably be more awkward for eZ80 code to use.

On balance, my inclination is to think that the pointer to a 4-byte value is probably the best approach.

sijnstra, 2025-04-12:
Some additional thoughts - is there any advantage in passing the data in a way that makes it more efficient for the internal routine to consume, in either speed or space? There's pros and cons to using either 24 bit or 16 bit registers. You have access to either instruction on the Agon in either mode so I don't think that's much of an advantage, but the (U) part is always annoying to handle.

In a way I'd like the approaches to be consistent so you can re-use ideas and code, macros etc.

stevesims, 2025-04-12:
I haven't done any analysis, mostly because I'm not familiar enough with cycle counts for instructions on the eZ80, but I would expect that having the APIs use a pointer would be marginally less efficient than the current two-register approach.  Code that makes use of these APIs however may be slightly simpler.  Overall it may still be marginally less efficient, but this would be by a relatively small number of cycles, and in the case of all the existing APIs these are APIs that are not commonly used.

Changing the two new FatFS APIs to use a pointer (#163) will make the APIs as a whole more consistent in how they handle such 32-bit values, especially if new variants of `mos_flseek` and `ffs_flseek` are added to support this approach.

In the specific case of the proposed three new `SD_*`APIs they would potentially be more commonly used APIs, but only in limited scenarios, such as a different OS or specialist tools requiring low-level SDCard access.

At some point in the future if/when we add an API to return C calling convention functions (see #137) then direct access to the SDCard functions can be provided, removing the need for a wrapper, and thus allowing code that wishes to use them to be more efficient than any potential MOS-style API.

My feeling is that overall the advantages of using a pointer to a 32-bit value outweigh the down-sides.

stevesims, 2025-04-12:
on reflection, it may actually be a better/simpler option to _not_ implement MOS APIs for these three SDCard functions, and instead to implement #137 specifically to expose the C calling convention SDCard functions directly

I would still plan on updating the `ffs_ftell` and `ffs_fsize` APIs #163 and add Z80 friendly versions of the two `*_flseek` APIs, as that will make the APIs overall more consistent in how 32-bit values are handled

stevesims, 2025-04-13:
as per notes added against #137 as Z80-mode and ADL-mode use two different stack pointers, the approach to exposing raw functions via that mechanism is not particularly simple.

therefore we will revert back to the original plan of implementing traditional MOS-style APIs for these three SD card functions

stevesims, 2025-04-14:
as these APIs are risky, the implementation in #168 uses a guard mechanism to prevent their accidental usage.  a program needs to get an unlock code, and then pass that unlock code to the other APIs.

at a later date the underlying functions will be exposed via a "get function" API (see #137) to allow them to be called directly (from ADL mode code).  using these will not require the use of the unlock code.

stevesims, 2025-04-21:
Implemented in #168 and #169

closed 2025-04-21 by stevesims